johnrickman/LiquidCrystal_I2C #1: Problems using the library within an interrupt

LennartHennigs, 2015-09-05:
Hey Marco,

I tried to use your library within an interrupt function but it seems to halt my Arduino (Leonardo).
For the interrupt handler I used the [TimerThree Llibrary] (http://playground.arduino.cc/Code/Timer1) with a default one second interval. The code worked before with a non I2C display and the "regular" [LiquidCrystal lib] (https://www.arduino.cc/en/Reference/LiquidCrystal).
My (uneducated) guess is that the delay() functions might cause some problems. 
Do you have any suggestions?
If you need more details or some code sample please let me know...

Cheers
l.

marcoschwartz, 2015-09-06:
Hello Lennart,

Unfortunately I am not the creator of the library, I simply forked it from: 

https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library

I did the fork to make it available within the Arduino IDE library manager. I'd suggest getting in touch with the owner of the library directly, but as you can see it's not maintained for about 4 years :(

LennartHennigs, 2015-09-07:
Hey Marco, 
thanks for the quick reply. I'll see what I can do.
I also found a thread discussing the problem in a bit more detail - might be of interest for others:
http://forum.arduino.cc/index.php?topic=140588.0

Cheers
l.

Avamander, 2015-10-18:
But this library is not a fork it's a reupload according to github. It would be nice if you'd actually fork it and give credit to the original author.