johnrickman/LiquidCrystal_I2C #7: Don't work correctly with Arduino 1.6.7 IDE Release

eipporko, 2016-01-10:
I just upgraded the IDE to the latest version (1.6.7), and I found that when I tried to print in the LCD, it shows only the first character of the string sended. 
So I decided to return to the 1.6.5 IDE Release. Any idea?

- Thanks

BoingNZ, 2016-01-14:
Same issue here

csamuelson, 2016-01-15:
Same issue

marcoschwartz, 2016-01-15:
Sorry to hear that! If somebody wants to take over this library I'm totally up for it, I didn't write it, just put it out there so it is in the library manager.

t3chguy, 2016-01-15:
In theory it should work, the return here (https://github.com/marcoschwartz/LiquidCrystal_I2C/blob/master/LiquidCrystal_I2C.cpp#L12) is what broke a lot of similar libraries.

sethiele, 2016-02-11:
I can confirm @t3chguy is right.

bperrybap, 2016-02-28:
Marco, you appear to have fixed this 4 months ago, but you need to re-tag it for the new 1.1.2 version.
This is very important as tags is how the Arduino library manager discovers library updates.
With out a tag, people will not get the update.
--- bill

marcoschwartz, 2016-03-04:
Done, let me know if it works :)

Bisotronic, 2016-03-16:
(I was waiting for other answers but...) @marcoschwartz it seems to be correctly fixed. Thanks.

jkalinovskis, 2016-04-07:
Working! THANK YOU!!!

alexbojko, 2016-08-13:
I'm using latest version of this lib (1.1.2) installed from library manager and it seem this bug is still reproducible for me.

scaytrase, 2016-08-14:
Same here, 1.6.9 + 1.1.2 (tried both from manager and custom zip from latest release)

highcastle1945, 2017-05-31:
I downloaded 1.1.2 from the Arduino library manager.  Still had issue.  I changed the LiquidCrystal_I2C.cpp file, lines 18-21 as below (changed function so it was not a void, and added the return 1 line):

'''#define printIIC(args)	Wire.send(args)
inline LiquidCrystal_I2C::write(uint8_t value) {
	send(value, Rs);
	return 1;
}'''

This seems to work. I was able to run the example code 'HelloWorld' in the library directory with some tweaks to fit my smaller 16x2 screen. Sorry if code style didn't turn out right in comment I am noob.

bperrybap, 2017-05-31:
Those lines you modified are for pre 1.0 IDE builds. write() cannot return a value in pre 1.0
Kind of looks like there may be some kind of issue with your build tools or core platform recipe file.

highcastle1945, 2017-05-31:
Better lucky than right I guess.  I will reinstall IDE and see if that works.

bperrybap, 2017-06-01:
I'd look at the compiler command line the IDE is using first. That will tell you what the value of ARDUINO is.

DavidStockford, 2019-04-10:
I am suffering from the same issue. I am running Arduino 1.8.9 IDE. I have installed the latest libray from Keyestudio. The CustomChar program shows that all the LED elements are working it appears to only be when trying to print a Character string that the problem occurs. Please can I have some guidance as I am new to this.

bperrybap, 2019-04-10:
It has been three years and this is still an issue. While the code was re tagged for vesion 1.1.3 but the library.properties was not properly updated and still says 1.1.2 which confuses the library manager. 
The library appears to have been assigned to a new person and from the readme, it now appears the project may be moving over to gitlab. I'm not sure if the Arduino IDE library manager works with gitlab.
My recommendation would be to switch to the hd44780 library and use the hd44780_I2Cexp i/o class.
https://github.com/duinoWitchery/hd44780
The library will "just work", as it auto detects the i2c address and the pin mapping used on the backpack so it will work with any of the backpack designs (where there are several) vs  just the one that is hard coded into LiquidCrystal_I2C
It is available in the IDE library manager so you can install it and be up and running in just a couple of minutes.
First run the included I2CexpDiag sketch to test everything, then you can run and look at the other hd44780_I2Cexp examples.
There is lots of additional information and documentation in an included "Documentation" sketch.

DavidStockford, 2019-04-10:
Thank you for your advice. I will get this library

darkworks, 2019-11-08:
> It has been three years and this is still an issue. While the code was re tagged for vesion 1.1.3 but the library.properties was not properly updated and still says 1.1.2 which confuses the library [...]

you lib seems complex , please make it simpler  easy and quick to use ... examples of your lib are scary :D