johnrickman/LiquidCrystal_I2C #9: Library Manager install of LiquidCrystal I2C 1.1.2 fails

per1234, 2016-04-25:
1. Sketch > Include Library > Manage Libraries > Filter your search... > search for LiquidCrystal I2C
2. Click on LiquidCrystal I2C(note the space rather than underscore between the words)
3. Install

Installation fails with either `CRC doesn't match. File is corrupted.` or `Error downloading http://downloads.arduino.cc/libraries/marcoschwartz/LiquidCrystal_I2C-1.1.2.zip` depending on Arduino IDE version. The cause of the issue can be found by looking at http://downloads.arduino.cc/libraries/library_index.json, which includes the following entries:
```JSON
    {
      "name": "LiquidCrystal_I2C",
      "version": "1.1.2",
      "author": "Frank de Brabander",
      "maintainer": "Marco Schwartz \u003cmarcolivier.schwartz@gmail.com\u003e",
      "sentence": "A library for I2C LCD displays.",
      "paragraph": "The library allows to control I2C displays with functions extremely similar to LiquidCrystal library. THIS LIBRARY MIGHT NOT BE COMPATIBLE WITH EXISTING SKETCHES.",
      "website": "https://github.com/marcoschwartz/LiquidCrystal_I2C",
      "category": "Display",
      "architectures": [
        "avr"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/marcoschwartz/LiquidCrystal_I2C-1.1.2.zip",
      "archiveFileName": "LiquidCrystal_I2C-1.1.2.zip",
      "size": 19856,
      "checksum": "SHA-256:7bcd8657a26644081bdfa9d8c9bb22937fcb2be08bb5aba02fff3756bc4291fb"
    },
```
and
```JSON
    {
      "name": "LiquidCrystal I2C",
      "version": "1.1.2",
      "author": "Frank de Brabander",
      "maintainer": "Marco Schwartz \u003cmarcolivier.schwartz@gmail.com\u003e",
      "sentence": "A library for I2C LCD displays.",
      "paragraph": "The library allows to control I2C displays with functions extremely similar to LiquidCrystal library. THIS LIBRARY MIGHT NOT BE COMPATIBLE WITH EXISTING SKETCHES.",
      "website": "https://github.com/marcoschwartz/LiquidCrystal_I2C",
      "category": "Display",
      "architectures": [
        "avr"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/marcoschwartz/LiquidCrystal_I2C-1.1.2.zip",
      "archiveFileName": "LiquidCrystal_I2C-1.1.2.zip",
      "size": 19858,
      "checksum": "SHA-256:1a8756bec354c95361efe68171eff341ae788a07b549709114134a8a5e475657"
    },
```
As you can see both entries have the same `url` value: http://downloads.arduino.cc/libraries/marcoschwartz/LiquidCrystal_I2C-1.1.2.zip. When it was created, the "LiquidCrystal_I2C" library file overwrote the "LiquidCrystal I2C" file so when you try to install LiquidCrystal I2C v1.1.2 the checksum doesn't match.

The cause of this issue is the incorrect `version` value in the library.properties of the 1.1.3 release: https://github.com/marcoschwartz/LiquidCrystal_I2C/blob/1.1.3/library.properties#L2. library_index.json is generated from the library.properties file of each release. The URL is created from the GitHub user name, library.properties name value(with spaces replaced with underscore), and the library.properties version value. So because you have different `name` values in library.properties from release to release(LiquidCrystal**_**I2C in [1.1.0](https://github.com/marcoschwartz/LiquidCrystal_I2C/blob/1.1.0/library.properties#L1) and [1.1.3](https://github.com/marcoschwartz/LiquidCrystal_I2C/blob/1.1.3/library.properties#L1), LiquidCrystal I2C in [1.1.1](https://github.com/marcoschwartz/LiquidCrystal_I2C/blob/1.1.1/library.properties#L1) and [1.1.2](https://github.com/marcoschwartz/LiquidCrystal_I2C/blob/1.1.2/library.properties#L1), this library actually shows as two different libraries in Library Manager but the filenames generated by the Arduino Library Manager system are equal..

The solution to this issue I propose is to delete release 1.1.3 because it has the incorrect library.properties `version` value and the only difference from release 1.1.2 is the changed library.properties `name` value(https://github.com/marcoschwartz/LiquidCrystal_I2C/compare/1.1.2...1.1.3).

**Deleting the tag on GitHub:**
- open https://github.com/marcoschwartz/LiquidCrystal_I2C/releases/tag/1.1
- click Delete
- click Delete this tag

**Deleting the tag in git:**
```
git tag -d 1.1.3
git push origin :refs/tags/1.1.3
```
After that you can make a new release 1.1.4 or 1.1.3-r2 with the correct `version` value in library.properties. Please choose a permanent library.properties `name` value and use that in all future releases to avoid causing further confusion.

Originally reported at: https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library/issues/3#issuecomment-210218409

bperrybap, 2017-08-08:
Seems like this may be an issue that should also be elevated to the actual library manager.
i.e. the library manager should not allow two different versions to use the same version number.
Its a pain for them to do the checks, but it seem like the library manager should not allow duplication version numbers.
Perhaps they could simply compare the tag version to the version number in the .properties file and if they don't match, don't accept the version.
They already do various sanity checks and drop versions for certain reasons, this would be just another sanity check.

pogo555, 2018-10-28:
So a solution has been proposed that would return this very useful library to use, but it hasn't been implemented. It is harder than per1234 says, or is it that if you fix it you own it, and nobody wants to step in that? 

I've got working code that uses this library and would like to update it. If I can't use this from the Arduino library vault, is it possible to download and compile the code?

per1234, 2024-04-30:
This problem was fixed via a cleaning of the Library Manager index 2016-07-22:

https://github.com/arduino/Arduino/issues/5058#issuecomment-227785970

The colliding `LiquidCrystal_I2C@1.1.2` release that was produced by @marcoschwartz pushing the 1.1.3 tag without first updating [the value of the `version` field in the `library.properties`](https://github.com/johnrickman/LiquidCrystal_I2C/blob/1.1.3/library.properties#L2) metadata file (https://github.com/johnrickman/LiquidCrystal_I2C/issues/18) was removed at that time.

The problem with the Library Manager system that allowed such collisions to occur was also resolved at that time.

closed 2024-04-30 by per1234