johnrickman/LiquidCrystal_I2C #18: tag 1.1.3 has wrong version in library.properties file

bperrybap, 2017-05-31:
The version number in the library.properties value is 1.1.2 for the 1.1.3 tag

BallscrewBob, 2017-08-08:
Second per1234's observation as it is also failing in CREATE on the "POSITIVE" keyword.

bperrybap, 2017-08-08:
POSITIVE keyword? This library does not have a POSITIVE keyword.

BallscrewBob, 2017-08-08:
My bad I am testing 6 i2c libs and opicked the wrong one...Am testing yours now with CREATE. 
sorry for confusion.

bperrybap, 2018-03-14:
Its been a long time and this is still an easily fixed issue that needs to be resolved.
The git tag and the .properties version must agree or things don't work correctly.

bperrybap, 2020-11-30:
The tags and library.properties file in this library are still broken.
This is a SERIOUS issue that needs to be fixed so users can download the library and install it using the IDE library manager.
AS IS, the versioning is all screwed up so the IDE library manager is confused and can't install the proper versions of the library.