johnrickman/LiquidCrystal_I2C #28: Using PROGMEM for custom chars - HOWTO

smartroad, 2018-09-29:
Hi All!

I thought I would add this here for anyone wanting to move any custom characters from RAM to ROM and save what limited space is available on the chip for more important things! I also want to say this is not my idea, but I got it from https://forum.arduino.cc/index.php?topic=519102.0 and modified it for this library.

in LiquidCrystal_I2C.cpp add the following (I added under the exisiting LiquidCrystal_I2C::creatChar):

```
void LiquidCrystal_I2C::createChar_P (uint8_t addr, const uint8_t *bitmap)
{
    uint8_t n;
    command (LCD_SETCGRAMADDR | ((addr % 8) * 8));
    for (n = 0; n < 8; n++) {
        write (pgm_read_byte (bitmap + n));
    }
}
```

then further down, under the section saying `// Alias functions` I added:

````
void LiquidCrystal_I2C::createChar_P (uint8_t addr, const char *bitmap)
{
    createChar_P (addr, (const uint8_t *)(bitmap));
}
````
Save and close LiquidCrystal_I2C.cpp then open LiquidCrystal_I2C.h
Under where it says `public:` add:

````
  void createChar_P (uint8_t, const char *);
  void createChar_P (uint8_t, const uint8_t *);
````
Save and close.
**Optional but useful**
Open the keywords.txt file and add in `CreateChar_P KEYWORD2` (the space between must be a TAB mark not a space) This will colour the text when you declare it in you sketch. Not essential but useful i think.

At this point you can open up in the Arduino IDE and use.

**EXAMPLE OF USE**

````
const byte face[8] PROGMEM = {
  B01110,
  B10001,
  B11011,
  B10001,
  B11011,
  B10101,
  B10001,
  B01110,
};
````

````
lcd.createChar_P(0, face);
lcd.write(0);
````

Hope this helps!!

modarsoos, 2020-02-07:
Good job

AgainPsychoX, 2021-05-26:
I wonder, why it's not in the base library? Can you make pull request set it as solution for this issue?