johnrickman/LiquidCrystal_I2C #32: ESP8266 Board Compatibility

Uwana, 2019-04-27:
Hello, please could you edit the library for compatibility issues with ESP8266 board, especially seeing that this library is used for the luftdaten.info project which is based on the ESP8266. Thank you.

hznupeter, 2020-03-19:
same problem,how to fix it?

paulwrath1223, 2021-05-18:
I'm new to github and arduino, but using LiquidCrystal_I2C lcd(0x27, 16, 2); causes an error on NodeMCU boards (and maybe others) but not on Mega 2560. The workaround is to replace it with LiquidCrystal_I2C lcd(0x27); and add lcd.begin(16,2); in setup. Maybe this is obvious, but it confused me a lot, and the error message was very unhelpful. ("Invalid Converson from ‘int’ to ‘t_backlighPol" [-fpermissive]")