johnrickman/LiquidCrystal_I2C #33: How do I use this to drive a LCD 20x4 ?

GeorgeFlorian, 2019-06-07:
Hello !

I am trying to drive a LCD 2004A. that has 20 cols and 4 rows.

I've tried `LiquidCrystal_I2C lcd(0x3F,20,4);` and `LiquidCrystal_I2C lcd(0x27,20,4);`.
But it doesn't work.
The backlight is on but that is all. No text, no nothing.

Do you have any clue ?

```
#include <Arduino.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x3F,20,4);  // set the LCD address to 0x27 for a 16 chars and 2 line display

void setup() {
  Serial.begin(115200);
  delay(500);
  
  lcd.init();                    // initialize the lcd 
  // Print a message to the LCD.
  lcd.backlight();
  lcd.setCursor(3,0);
  lcd.print("Hello, world!");
  lcd.setCursor(2,1);
  lcd.print("Ywrobot Arduino!");
  lcd.setCursor(0,2);
  lcd.print("Arduino LCM IIC 2004");
  lcd.setCursor(2,3);
  lcd.print("Power By Ec-yuan!");
}

void loop()
{
}
```

iFrostizz, 2019-06-07:
Try to use a I2C scanner and write your I2C address at the place of 0x3F

GeorgeFlorian, 2019-06-07:
> Try to use a I2C scanner and write your I2C address at the place of 0x3F

I did, and it is 0x27 and this `LiquidCrystal_I2C lcd(0x3F,20,4);` doesn't work.
It outputs gibberish:
[image]

iFrostizz, 2019-06-07:
Which Arduino model ?

GeorgeFlorian, 2019-06-07:
> Which Arduino model ?

ESP32, actually ! DevKit V4 Wrover-B.

But it works now !

It turns out that the issue came from the pins not making proper contact.
After I've soldered them it works wonders !

closed 2019-06-07 by GeorgeFlorian

iFrostizz, 2019-06-07:
Nice ! :)