johnrickman/LiquidCrystal_I2C #37: i2c is not allowed ?

hasanemretonguc, 2020-03-16:
Code working just fine but gives me this error:
 object of abstract class type "LiquidCrystal_I2C" is not allowed: -- pure virtual function "Print::write(uint8_t)" has no overrider
[image]

Brock13s, 2020-04-09:
Hey there hasanemretonguc I have found a solution on the internet and that was to open the library folder and there should be LiquidCrystal_I2C.h and LiquidCrystal_I2C.cpp now open the both files in vscode which you are using and then in the header file on line 84 where it says `virtual void write(uint8_t);` change the void to size_t then once you have that done go into the cpp file and on line 19 `inline void LiquidCrystal_I2C::write(uint8_t value)` change the void to size_t like the header file and that should work for you since I had the same problem as you and I tried the solution on the internet and now it works :)

hasanemretonguc, 2020-04-11:
thank you that fixed my issue

closed 2020-04-11 by hasanemretonguc