johnrickman/LiquidCrystal_I2C #38: I2C hangs / freezing

diesel52, 2020-05-07:
Hi,

Firstly, thank you a lot for this useful library. Nice work. 

I have use a lot your library for different project, with good results.
But for two of them, who are using a long cable beetween arduino and I2C lcd, i have frequently some garbage on screen, or some freezing from arduino.

All my search about I2C crashs let me think that when we are near the capacitance limit of I2C bus (400pf ?) I2C can freeze. 
There is few solution, one of them consist to replace Wire.h by I2C.h, because this last one permit to set a timeout function who avoid freezing when I2C dont read a complete answer from slave. 

There is not this function in your library, do you think that could be an evolution ? 
I know that i'm not alone in this case about I2C bus (in general), but i have no idea about how many people are impacted with liquidcrystal. 

Thanks

stefschin, 2020-10-20:
With STM32 Blue Pill, sometimes gets display freeze but Microcontroller continues to work, so I guess it's a communication problem

doppelhub, 2021-08-13:
@diesel52 is correct that this library hangs forever if the data line stays low at the end of transmission (e.g. caused by too much capacitance on I2C data line).  

One solution is to add stronger pullups near the I2C master... I recommend 10k.

But yes, this library should timeout gracefully, but right now it just hangs forever.

Steps to reproduce:
0: Setup hardware to blink an LED each time through a loop
1: Begin serial transmission to screen... maybe send a counter (e.g. 1, 2, 3, 4, 5, etc), so you can see the screen updating.
2: Place pulldown resistor from I2C data line to ground.  Result: Arduino hangs.
3: Remove pulldown resistor.  Result: Arduino either returns (and counter keeps incrementing)... or it continues to hang.