johnrickman/LiquidCrystal_I2C #44: Create a license file

bperrybap, 2020-07-23:
johnrickman,
Please add a license file to this repository so that people know the true license of this work.
The license must be LGPL 2.1+

This code came from code that has always been and must be licensed as LGPL 2.1+
This code came from the LiqudCrystal_I2C github project:
https://github.com/marcoschwartz/LiquidCrystal_I2C
https://github.com/johnrickman/LiquidCrystal_I2C
but the code was originally created by Mario H back in 2009 and was derived from Arduino.cc LiquidCrystal sources.
https://hmario.home.xs4all.nl/arduino/LiquidCrystal_I2C/
More details below:
While the IDE LiquidCrystal code did not contain copyright notices or license
information in the LiquidCrystal source files themselves,
the IDE LiquidCrystal code was licensed as LGPL 2.1 as indicated in the
license.txt file in the root directory of the arduino.cc IDE repository which
was committed October 11, 2007 and has not changed since.
It can be seen here:
https://github.com/arduino/Arduino/blob/master/license.txt
license.txt was a catchall license for all
Arduino core files and libraries provided by and bundled with the IDE.

It contains a small note:
```
next, the gnu lesser general public license that covers the arduino core
and libraries.
```

What is below that tiny comment in the readme file is a LGPL 2.1 license agreement.
So it seems that the intent was that the LiquidCrystal library was released as LGPL 2.1

In 2015 the IDE LiquidCrystal source files were updated to contain a copyright notice and
license within the files.
```
Copyright (C) 2006-2008 Hans-Christoph Steiner. All rights reserved.
Copyright (c) 2010 Arduino LLC. All right reserved.

[ with a LGPL 2.1+ notice ] that clarifies that the license is LGPL 2.1+
```

In 2015 the IDE LiquidCrystal source files were updated to contain a copyright notice and
license within the files.
```
Copyright (C) 2006-2008 Hans-Christoph Steiner. All rights reserved.
Copyright (c) 2010 Arduino LLC. All right reserved.

[ with a LGPL 2.1+ notice ]  that clarifies that the license is LGPL 2.1+
```

francesco-scar, 2021-04-03:
I would like to edit this library for my own use, in the current state of this project can I do that? And if I publish my edited version which license I have to use?

bperrybap, 2021-04-03:
As I showed, the license for this library is LGPL 2.1+
Your derivative must conform to that license. 
That means you must use the terms of the LGPL 2.1 license.
It states that the license is LGPL 2.1 or (at your option) any newer version.
Which means you can use LGPL 2.1 or LGPL 3.0 or do a one time conversion to GPL 3.0

But you cannot change it to something like BSD, MIT, Apache, etc... or any other non GNU license as those are not allowed by the existing LGPL 2.1 license.