johnrickman/LiquidCrystal_I2C #47: GitLab at https://gitlab.com/tandembyte/LCD_I2C no longer available

bperrybap, 2020-10-30:
https://gitlab.com/tandembyte/LCD_I2C
as referenced on the main repository readme is no longer available.

Fenytestver, 2020-12-20:
Looked for the same. Thanks for the note. 
Is this repository still maintained?

bperrybap, 2020-12-20:
I would say no. 
The repository was never maintained that well as the person who initially created it just created it to get it into the IDE library manager. He didn't have the skills to maintain it and there have been several easy to fix issues that have lagged for several years.
One issue is serious due to git tags being incorrectly created that confuses the IDE library manager's ability to get the proper versions.

It was handed over to johnrickman somewhat recently.
He basically copied it over to gitlab, changed its name a few times, then changed the license to BSD, which is not allowed under the LGPL terms and then abandoned this repository.
I pointed out the licensing issues, and he originally seemed open to correcting it.
But shortly after that, he deleted the gitlab repository.

At this point, he seems to have removed the repo on gitlab and abandoned his tandembyte account and hasn't been responding to issues on this repository.
Not sure how to even contact him at this point.

If you are looking for an actively maintained full featured library that supports hd44780 LCDs using a PCF8574 based backpack, you may want to consider the hd44780 library using the hd44780_I2Cexp i/o class.
You can install directly from the IDE GUI using the IDE library manager. (not from a zip file)
https://github.com/duinoWitchery/hd44780

bperrybap, 2020-12-21:
I found a contact email and sent johnrickman an email and asked him about this repository and the one on gitlab.

Alceatraz, 2021-01-05:
> I would say no.
> The repository was never maintained that well as the person who initially created it just created it to get it into the IDE library manager. He didn't have the skills to maintain it [...]

OFF TOPIC: Sounds awful, Will you call this some bad or even worse behavior?

bperrybap, 2021-01-05:
Still no response from the email I sent to johnrickman yet.
It hasn't bounced in two weeks so I'm assuming the email account still exists.
I sent it with return receipt but no response yet.

CollinCampbell, 2022-05-03:
Forked repo: https://gitlab.com/joearmstrong980/LCD_I2C

bperrybap, 2022-05-03:
Any idea what happened to John Rickman and his tamdembyte  LCD_I2C repo on gitlab??
He took over this repo, moved it to the tandembyte gitlab repo , renamed it LCD_I2C and then after illegally changing the license from LGPL to MIT the tandembyte LCD_I2C repo becomes inaccessible.
Now it shows up again on gitlab as a fork in joearmstron980
I'll create an issue in joearmstrong980's repo to see if I can at least get him to fix the license there.

drf5n, 2023-03-16:
> If you are looking for an actively maintained full featured library that supports hd44780 LCDs using a PCF8574 based backpack, you may want to consider the hd44780 library using the hd44780_I2Cexp [...]

The hd44780 library from the Arduino IDE's Library Manager can replace a LiquidCrystal_I2C setup with a few differences.  The main differences are that instead of setting the chars and lines in the global declaration and then using `lcd.init()` in the setup, you set the chars and lines inside setup with `lcd.begin(chars,lines);`

Here's a DFRobots/Frank de Brabander [LiquidCrystal_I2C helloWorld.ino](https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library/blob/master/examples/HelloWorld/HelloWorld.ino) script modified to work with hd44780: 

```
#include <Wire.h>
//#include <LiquidCrystal_I2C.h> // https://github.com/johnrickman/LiquidCrystal_I2C
#include <hd44780.h>                       // main hd44780 header https://github.com/duinoWitchery/hd44780
#include <hd44780ioClass/hd44780_I2Cexp.h> // i2c expander i/o class header

// Set the LCD address to 0x27 for a 16 chars and 2 line display
//LiquidCrystal_I2C lcd(0x27,16, 2);

hd44780_I2Cexp lcd; // declare lcd object: auto locate & auto config expander chip

void setup()
{
  // initialize the LCD
  //lcd.init();
  lcd.begin(16,2);

  // Turn on the blacklight and print a message.
  lcd.backlight();
  lcd.print("Hello, world!");
}

void loop()
{
  // Do nothing here...
}
```

The DF Robots code is available from https://www.dfrobot.com/product-135.html

bperrybap, 2023-03-16:
The hd44780 library wiki has information about porting code using other LCD libraries to using hd44780 hd44780_I2Cexp i/o class:
https://github.com/duinoWitchery/hd44780/wiki/ioClass:-hd44780_I2Cexp
See the "Conversion" section:
https://github.com/duinoWitchery/hd44780/wiki/ioClass:-hd44780_I2Cexp#conversion-from-other-libraries

--- bill

per1234, 2024-04-30:
Duplicate of https://github.com/johnrickman/LiquidCrystal_I2C/issues/39