johnrickman/LiquidCrystal_I2C #55: Random Characters displays ESP32

Sladerix, 2021-12-06:
Hi everyone, i use an ESP32 wired up to a LCD 16x2 I2C. I use the esp32 for the BLE.
Unfortunately some time random characters displays...

https://user-images.githubusercontent.com/6717214/144840296-eb33b49c-a128-46d7-9906-25c27605fecf.mp4

I already tried with this solution: https://www.youtube.com/watch?v=icdEifNHzeY&ab_channel=AndreasSpiess
by puttin Wire.setClock(10000); after the intialization of the lcd, but the problem is facing again. What could it be?

Thanks in advance

wongzhiwei98, 2021-12-17:
I have the exact same issue. For me, the weird character doesn't come out immediately, it happens after a few minutes. But it always happens and causes the ESP32 to crash eventually.

Sladerix, 2021-12-17:
> I have the exact same issue. For me, the weird character doesn't come out immediately, it happens after a few minutes. But it always happens and causes the ESP32 to crash eventually.

Did you find any solution? Maybe with another library? I’m struggling a lot for this annoying issue

wongzhiwei98, 2021-12-18:
I have just tried with another library by "enjoyneering/LiquidCrystal_I2C". Although the character still goes haywire after some time, at least my other task still working properly. And it seems to be is it more stable and doesn't crash my ESP32 anymore.

Sladerix, 2021-12-18:
> I have just tried with another library by "enjoyneering/LiquidCrystal_I2C". Although the character still goes haywire after some time, at least my other task still working properly. And it seems to be is it more stable and doesn't crash my ESP32 anymore.

Thanks for the suggestion!
Later i will take a look, meanwhile i've a question: the library has the same methods? is sufficient to replace the library without rewrite the code?

wongzhiwei98, 2021-12-18:
Mostly the same but the initialization of the LCD is different. I would suggest you read up on their repository [here](https://github.com/enjoyneering/LiquidCrystal_I2C) to look at their example. I am still experimenting with this library currently. I let you know if I have successfully implemented using this library by enjoyneering.

wongzhiwei98, 2021-12-19:
I have successfully implemented using the Liquid Crystal library by enjoyneering. However, it is still not perfect. There are still glitches to the LCD from time to time, however, it doesn't affect the rest of the code. Here are the tips you should try to reduce the glitching of LCD. Reinitialize the LCD at a certain interval. This is will remove and refresh your I2C display. Here are my example code.

```
  //Reinitialize the LCD every 9 seconds because library is sensitivity to signal noise in the circuit.
  if((long)(millis() - last_millis_lcd) >= 9000)
  {
    lcd.begin(16, 2);
    last_millis_lcd = millis();
  }
```
As I am using ESP32, the library doesn't work out of the box. You have to comment out on one of the lines in the library to disable the analog write. This line is located in the libdeps/esp32/LiquidCrystal_I2C/LiquidCrystal_I2C.cpp  ,line 951.

```
void LiquidCrystal_I2C::setBrightness(uint8_t pin, uint8_t value, backlightPolarity polarity)
{
  pinMode(pin, OUTPUT);

  if (polarity == NEGATIVE) value = 255 - value;

  //analogWrite(pin, value); //COMMENT OUT THIS LINE TO WORK
}
```

Sladerix, 2021-12-23:
> I have successfully implemented using the Liquid Crystal library by enjoyneering. However, it is still not perfect. There are still glitches to the LCD from time to time, however, it doesn't affect [...]

Thanks very much for sharing this. This problem is not very discussed on web, i think.

Sorry for not replying, but i'm waiting for the problematic hardware to comes to me in order to doing the tests... but actually what i'm thinking is that if there is this type of unstability I'll probably opt for a classic parallel lcd to solve any issue (i hope)