johnrickman/LiquidCrystal_I2C #82: Gitlab link is broken

cmdrscotty, 2024-11-13:
Link on this github repo directing everyone to gitlab for documentation is broken.

This needs to be fixed as this repo is referenced in multiple sources as the go to for the library documentation.

per1234, 2025-01-15:
Duplicate of https://github.com/johnrickman/LiquidCrystal_I2C/issues/39

rtallarr, 2025-07-24:
did you find a working link somewhere?

cmdrscotty, 2025-07-24:
> did you find a working link somewhere?

I ended up just using https://github.com/duinoWitchery/hd44780

According to https://github.com/johnrickman/LiquidCrystal_I2C/issues/47 the author of this repo basically took the code, locked it behind a BSD license, then subsequently just deleted everything and abandoned this library all together, leaving it in the state it is while many upon many Arduino examples and official resources are still linking back to this now dead repo.

Someone did also fork this repo and make some updates to it as well: https://github.com/enjoyneering/LiquidCrystal_I2C